ARM-software/CMSIS_4 #106: osMessageCreate will return NULL when enable DMBED_HEAP_STATS_ENABLED

kyliuxing, 2017-02-22:
mbed OS version : 5.1.5
compile cmd: mbed compile --source TESTS/TARGET_RDA/test/ --source ./ -DMBED_HEAP_STATS_ENABLED

I make a wrapper of osMessageCreate, when i define MBED_HEAP_STATS_ENABLED，osMessageCreate will return NULL. Is it a bug and has been fix in latest mbed version?

test case:
main.cpp
#include "mbed.h"
#include "rtos.h"

#define osWaitForever 0xFFFFFFFF
void* rda_msgQ_create(unsigned int queuesz)
{
	void* internal_data;
	osMessageQDef_t msgQ;
	osMessageQId msgQId;

	internal_data = (void *)malloc((4 + queuesz) * sizeof(unsigned int));**//Is it the root cause?**
	msgQ.queue_sz = queuesz;
	msgQ.pool = internal_data;
	msgQId = osMessageCreate(&msgQ, NULL);

    printf("msgQ addr %p\r\n", msgQId);
	return (void *)msgQId;
}

int main() {
	void *msgQ;
	msgQ = rda_msgQ_create(4);
    while (true) {
    }
}
rt_cmsis.c
osMessageQId svcMessageCreate (const osMessageQDef_t *queue_def, osThreadId thread_id) {

  if ((queue_def == NULL) ||
      (queue_def->queue_sz == 0U) ||
      (queue_def->pool == NULL)) {
    sysThreadError(osErrorParameter);
    return NULL;
  }
  
  if (((P_MCB)queue_def->pool)->cb_type != 0U) {
    sysThreadError(osErrorParameter);
    return NULL;
  }

  rt_mbx_init(queue_def->pool, (uint16_t)(4U*(queue_def->queue_sz + 4U)));
  mbed_error_printf("\r\nsvc msg addr %p\r\n", queue_def->pool);**//I add this to debug**
  return queue_def->pool;
}
Result:
MBED_HEAP_STATS_ENABLED is defined
    svc msg addr 00100224
    msgQ addr 00000000
MBED_HEAP_STATS_ENABLED is not defined
   svc msg addr 001001f4 //it seems have two msg Queue
   svc msg addr 00180018
   msgQ addr 00180018

0xc0170, 2017-02-22:
Hello, can you report this to https://github.com/ARMmbed/mbed-os, where we can have a look at it. If its bug in cmsis, we then could report it here.

kyliuxing, 2017-02-23:
@0xc0170 
I have report this to mbed-os, thank you!
https://github.com/ARMmbed/mbed-os/issues/3825

bulislaw, 2017-02-23:
You need to zero the memory as as mentioned here https://github.com/ARMmbed/mbed-os/issues/3825#issuecomment-281967483