ARM-software/CMSIS_4 #109: Address of SCS in CMSIS not defined in ARM Cortex-M7 Generic User Guide

mguntli, 2017-06-21:
# Summary
There is an issue regarding a documentation mismatch about the synonym *SCS (System Control Space)*.
But if even the simplest base addresses do not match between CMSIS and the ARM Cotrex-M7 Generic User Guide, what should be the reference then?

# Problem
The SCS_BASE address is used in CMSIS, but the synonym "SCS" does not exist in the ARM Cortex-M7 Generic User Guide (ARM DUI 0646B).
Of course I know that the reason that CMSIS the SCB does not match to the ARM User Guide is to not blow up the entire structure with 3320 reserved bytes, but it still is implemented in a very dirty way and the naming is a quite a mess..

# Core_CM7.h
https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/core_cm7.h
* SCS is defined at 0xE000E000
https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/core_cm7.h#L1741
* SCB is defined at 0xE000ED00
https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/core_cm7.h#L1748
* SCnSCB is defined at 0xE000E000
https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/core_cm7.h#L1750

# ARM DUI 0646B:
http://infocenter.arm.com/help/topic/com.arm.doc.dui0646b/DUI0646B_cortex_m7_dgug.pdf
* A search for "System Control Space" returns 3 matches, but there is no address defined
* A search for "SCS" returns 2 matches, but there is no address defined
* Table 4-1 states that the System control block address range is 0xE000E008-0xE000E00F
* Table 4-12 lists 0xE000E008 as ACTLR (the first register of System control block)

jkrech, 2017-06-21:
Version 4 is being superseded by Version 5
Please review CMSIS_5 here: https://github.com/ARM-software/CMSIS_5

mguntli, 2017-06-21:
Issue still exist in CMSIS Version 5, created a new issue there (https://github.com/ARM-software/CMSIS_5/issues/207)

closed 2017-06-21 by mguntli