ARM-software/CMSIS_4 #116: CMSIS-SVD.xsd accepts XML files that were probably not intented to be valid

gralpli, 2018-02-02:
In the file _CMSIS/Utilities/CMSIS-SVD.xsd_ on lines 175–179 there is the following definition:
```xml
<xs:simpleType name="enumeratedValueDataType">
  <xs:restriction base="xs:string">
    <xs:pattern value="[+]?(0x|0X|#)?[0-9a-fxA-FX]+"/>
  </xs:restriction>
</xs:simpleType>
```
which also accepts wrong values such as `#7F` (numbers bigger than one and letters in a binary literal) and `0b0010`.

I found the latter one in an (valid) XML file from Infineon with the intention of being a binary literal. Based on the `(0x|0X|#)?` part of the regex I assume that such a syntax is not allowed (and `#0010` should have been used instead) and that the file validated just by accident.

jkrech, 2018-02-05:
Good catch. Did you actually find any tools failing to support 0b as binary indicator? The SVDConv shipped at the moment accepted it as far as I can see.
The documentation needs to be improved as well.
Options:
all formats have an optional leading '+'
 
1. decimal number [0-9]+
2. hexadecimal number with leading 0x or 0X 
3. binary with leading 0b0 or #

The 'don't care' is only supported in the binary (3rd) format and an x or X is used for that:
e.g. 0b10x0 or #010x0

Please review the pattern below if you see any omissions or issues with it.
```xml
  <xs:simpleType name="enumeratedValueDataType">
    <xs:restriction base="xs:string">
      <xs:pattern value="[+]?(((0x|0X)[0-9a-fA-F]+)|([0-9]+)|((#|0b)[01xX]+))"/>
    </xs:restriction>
  </xs:simpleType>
```
Thanks.

gralpli, 2018-02-06:
Yes, this looks good.

[svd2rust](https://github.com/japaric/svd2rust) did't support it, but it's a small tool that is easy to change.